bigtreetech/BIGTREETECH-TouchScreenHardware #1: Connector type for UART ports on TFT35 E3 v3.0

prjoni99, 2020-07-24:
Hello,

I'm trying to find out what connector to get to connect a raspberry pi to the uart port on the back of the TFT35 display. Right now I have DuPont cables but I want to have a proper connector for this. 

Thanks

traffic-light, 2020-08-08:
Jst xh

traffic-light, 2020-08-08:
I bought these for JST:
[Connectors](https://www.banggood.com/150pcs-234Pin-JST-XH-2_54mm-Dupont-Connector-MaleFemale-Wire-Cable-Jumper-Pin-Header-Housing-Connector-Terminal-Kit-p-1447810.html?rmmds=myorder&cur_warehouse=CN)
[Crimpingtool](https://www.banggood.com/SN-01BM-Terminal-Wire-Cable-Crimping-Pliers-Tool-for-Dupont-PH2-0-XH2-54-AMG28-20-KF2510-JST-p-1301387.html?rmmds=myorder&cur_warehouse=CN)

oldman4U, 2020-09-07:
@prjoni99 

Please do not forget to close this ticket once you do not need it anymore.

Thank you

closed 2020-09-07 by prjoni99