bigtreetech/BIGTREETECH-TouchScreenHardware #2: TFT70 v3

MrCHarley, 2020-08-19:
Just hooked up to SKR E3 mini 2.0 with a TFT70 v3 (fw 3.0.26.2). 
One thing i have noticed is it does not like you to change the extrusion  temperature "On the fly" in Marlin UI. I tried to decrease  the temperature, when i did, the printer stopped for a couple of seconds, then inched a bit more, stopped... an so on... With the TFT35 which i used until now, that was not an issue.
The TFT was just swapped with the original connection for the TFT35

oldman4U, 2020-09-07:
Hi.

Which firmware are you using? There has been an issue where the printer stopped working after changing the temp or flow value resulting in a flickering value. The latest firmware does not have this issue.

Hope this helps

BTW. I would start tickets only in the BTT_Firmware repository which has many more active users.

bedilaci, 2021-01-03:
Hi,
i have a SKR 1.4 +TMC2008UART+ Geeetech 3D touh + BTT TFT70 V3. My problem: when i use in UI mode, the printer randomly Stop. **12864 simulator mode perfect** !
Marlin 2.0.7.2, TFT70 firmware upgrade to latest, I monitoring with M115 and run g code in send information, and when stopped, then is a simple   **Send:OK**     or something meaningless
Thanks and regards
Laszlo
[image]
[image]