bigtreetech/BIGTREETECH-TouchScreenHardware #7: BTT tft 7.0 not working

fke1964, 2021-01-18:
Goodmorning 
i have an skr 1.4 pro mounted in my CR20pro . I have attached an BTT TFT7.0 screen all new. I am getting only a white screen.
It looks like i don't get a signal from the board. I tested with the old screen and that works so marlin is installed in the board. I tried also different cables but no luck. It looks like an Marlin issue. what do i have to do to get the RS232 port working and also the other two. i have #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER uncommented.
can you help me out here?

aeaoeiao, 2021-03-21:
I have the same issue...

fke1964, 2021-03-21:
Well the problem was easy. Pull out the BL_touch and the screen works. The BL-touch was shortning the board. So I changed the colors and it works