bigtreetech/BIGTREETECH-TouchScreenHardware #14: Pi TFT50 V2.0 DPMS Support?

darouss99, 2021-12-15:
Hi, 

Is the Pi TFT50 V2.0 support DPMS to disable the backlight? There is a wheel to play with the brightness.  But I want to disable the backlight completly automaticly at sleep.

Thank you

eduncan911, 2022-10-18:
So, we brought this up before in the old archived repo: https://github.com/bigtreetech/OctoBTT/issues/12

In short, there is no I/O (software) way to control the backlight.  It's only via the wheel.

However, BTT support did respond to that old issue above saying that you can remove `R6 0 Ω` resistor and place a reostate/switch/etc there instead.

Another user said it's a simple PWM 3.3V signal that you can configure from an RPi pin to solder in place of the R6 connector.

I am about to do this myself in the next week or two.

However, my problem is, I have no idea which resistor is "`R6`" - nothing is labelled on the board (well, not much) and the posted schematics do not show an R6.  Only a couple components are identified, but nothing else.

Eclypse3D, 2022-11-10:
It is very odd that in 2022 they are putting out a LCD display that cannot run DPMS, turn off the backlight, and wake up out of sleep.  Come on BTT put some effort into it at least, the excuses "at this time we cannot do this" is not a valid response.  Right now on my 50 v2 it will go to sleep, and stay white, but it also does not come out of sleep when you tap the screen.  you have to literally reset klipper screen, reboot or make a change in the interface to get it to re-display.  Yet sitting right next to my Voron is a 7" RPI screen with no display rocker that can go to sleep and wake up and shut off the display, and it runs a ribbon cable interface just like the BTT TFT50 V2.......  there is a reason why its a $48 display....

functionpointer, 2023-02-09:
@Eclypse3D which display are you using with working sleep?

arekm, 2023-11-09:
There is Pi TFT50 V2.**1**. Does anyone know it that version has this fixed or not?

arekm, 2024-01-08:
[image]

Got v2.1 - it no longer uses potentiometer. Now there are two buttons for brightness. Also DPMS is working fine. Backlight can be turned off via DPMS.

hololight, 2024-07-08:
Does anyone know if I can flash the newer firmware to the older version to get the DPMS support? I'd really like to be able to shut the screen off...