bigtreetech/BIGTREETECH-TouchScreenHardware #26: PITFT70 Touch screen no longer responding

Macaiah, 2023-03-14:
Ran into this issue 2 days ago. My PITFT70 is no longer responding to touch input. Displays all graphical information, just no touch response. This happened after updating Klipper and a couple of the bundled modules.

Anyone have any ideas how to restore this functionality?

cinusik, 2023-03-15:
Same here with two PITFT50 - touch screen not responding. On one of them  "touch" died 5 days ago, on the second one it happened today. Same question here. Looks like Klipper update bricked "touch" functionality. Already sent one for replacement but I'm pretty sure it will happen again if I connect replaced unit. I also relate it to last Klipper/RPI update.

cinusik, 2023-03-15:
I found the solution. Looks like Klipper update changed something regarding to RPi config.
Run:
`sudo raspi-config`
and enable i2c in Interface Options.
`reboot`

Macaiah, 2023-03-15:
Thanks. I figured it would be an i2c problem, just had more pressing matters to tend to on a new VZBot build. I'll give it a shot when I get home.

closed 2023-03-21 by Macaiah