bigtreetech/BIGTREETECH-TouchScreenHardware #32: Please provide the CAD Files for BTT Pi TFT50 V2.1

mneuhaus, 2023-06-14:
Currently i can only find v2.0 but not v2.1 which changed the brightness wheel to buttons which conflict with some existing mounting CAD files, please provide the CAD files for v2.1

FriedSemiconductors, 2023-07-10:
Yes, that would be awesome.
I'd also appreciate, if the CAD and Documents of the V2.1 would be added.
The official shop pages for the V2.1 even link to this github repo, just to find out, it is not uptodate.

[image]

KiwiChrist, 2023-07-25:
@mneuhaus I totally agree. Would be good for BTT to supply an up to date stp/stl file of the unit and all the cad drawings showing the thickness as well, not just a plan view. 

I will draw up a model of the BTT PITFT50 once I have reeceived it (end of July 2023). It'll be dimensionally accurate but pretty simple model.  I note that downloading the v2.0 stp file, which is not what I have purchased is incomplete, no screen modelled nor all of the connectors, just the pcb so that's a waste of time. Even if the v2.1 was available I'd say the stp would be incomplete too.

LukasExemplar, 2024-10-11:
I'd like to echo your sentiment. It seems like a big oversight! This screen is highly popular for KlipperScreen users!