bigtreetech/BIGTREETECH-TouchScreenHardware #33: Can you provide the updated CAD files

fizzystech, 2023-09-23:
I have designed an enclosure for the BTT Pi TFT50 V2.0 and there are some changes with V2.1 in that you removed the dial and replaced it with buttons. Can you guys upload a step file of the latest board. Also the previous step file does not load up correctly so I hope you can fix all these issues as it make it more difficult to design really good tight tolerance enclosures.

Here is my current design for all of the BTT Pi enclosures: -
BTT Pi TFT50 - https://www.printables.com/model/405237-voron-switchwire24-btt-pi-tft50-screen-mount
BTT Pi TFT70 - https://www.printables.com/model/581227-ft-btt-pi-tft70-screen-mount
BTT Pi TFT43 - https://www.printables.com/model/405160-voron-switchwire24-btt-pi-tft43-screen-mount

LukasExemplar, 2024-12-05:
I second this!

neptunus0001, 2026-09-02:
@bigtreetech Why is it taking so long for the STEP to become available?